GW8RDI/uSDXOpen #3: GPLv3 violation.

cathalferris, 2023-05-23:
This repository is in violation of the GPLv3 license.

Because you list the binaries as being released under GPLv3, you absolutely must have the source code available that creates the binaries that you make available - even if the code you used was itself open.

Not making the source code directly available is a GPLv3 violation. You do not get to enforce GPLv3 on others without fully respecting the license yourself.

To resolve this issue, you must put all of the source files up that were used to compile to the binaries that are listed here.

GW8RDI, 2023-05-23:
Dear Mr Cathal Ferris, our lawyers have already tried to contact you and your employer regarding harassment.  They will continue to seek your address and take legal action against you.
Your claims are senseless, this GitHub page only contains source code.

cathalferris, 2023-05-23:
:innocent:

Oh that's *so* cute. 
<pats-on-head>

--
Given your well-known recent mistakes with license interpretation, please make sure your "lawyers" :wink: have read and can understand [](https://www.gnu.org/licenses/gpl-faq.en.html) to make sure you don't "accidentally" fall foul when again not abiding by licenses.

GW8RDI, 2023-05-23:
Cathal Ferris for a man of over 50 years of age, you are acting very immaturely.  I don't know what your problem is, but if you don't stop wasting our time you'll be blocked from this page.

cathalferris, 2023-05-23:
It's good that you can admit you don't know what my problem is, because it's as simple as my having no problem at all. 
It is good that you can see that at least.

I'm not really sure why you would think I would have a problem, with you in particular anyway, as you're apparently not abusing licenses on my code at this point at least.

You do however appear to have a highly inappropriate level of antagonistic reactions to certain situations. Why *is* that? Conflict syndrome? High-conflict-personality? Both? 

*I* really am not wasting *your* time here.  
I'm not sure who you think the "our" in your statement refers to - I'm certainly not wasting my time at least.
If you're unwilling to lean from your mistakes, then you are probably wasting your time on more than this endeavour, but that's not on me.  
You don't have any contributors, so I couldn't possibly be wasting anybody else's time here.

As for me, I just like attempting to educate those that don't understand how licensing (especially the various flavours of Creative Commons and other Open Source licenses) works in the real world. It does result sometimes that those that abuse those licenses get overly-defensive through their misunderstandings of  things.

If you want to block me for having taken my valuable time to directly help you not make another public mistake regarding your use of licensed code, then that will have to be seen as abuse - by you. I've not given you a valid reason to block anyone after all.

Anyway, I hope you have a good licencing-compliant day, and I hope that it doesn't rain on you too much down there.
(here's a free Github hint - you can probably close this issue with a "wont-fix"...)

chrisdebian, 2024-08-02:
Tidying up, can this be closed, now?

Chris
2E0FRU

cathalferris, 2024-08-03:
No, I don't think this can be closed, as the situation I opened it for is not actually resolved.
The attempts of this fork to meet the license terms aren't actually meeting the license terms.

The readme of this project is also incredibly dishonest at the moment. Guido's text from the forked-from project does not link to the closed-group forums on Facebook that the originator of this fork rules with an iron fist. Guido's project links here: [uSDX Forum](https://groups.io/g/ucx). Maybe it would be correct to remove the edited text and instead have Guido's actual text instead of putting words in Guido's mouth..

This project still implies that this fork can be used on Manuel's project - but it is not licensed, and this is not addressed in this fork's documentation.

In short, this bug will be re-opened if closed, until the license terms are properly adhered to, and the documentation properly shows this in a clear and non-passive-aggressive manner..

(Also - it's not smart to threaten legal routes, especially if there's no follow-up. I was looking forwards to smacking any frivolous lawsuit back to where it belonged. The lack of a follow-up of any type does lend credence to my points. Such wonderful interactions with the community by the originator of this fork of Guido's code.)

GW8RDI, 2024-08-03:
Cathal, one day I am going to meet you, and we can have a real discussion.
EU and US law is very clear in that owners of a product have the right to repair and maintain their own products exactly as they choose.  Legal cases where companies tried to force users to pay for their support were all thrown out.  Anyone interested can google "law right to repair".  You of course are not interested in people's rights.

cathalferris, 2024-08-03:
Well isn’t that an interesting reply from the maintainer of this fork of Guido’s code?
Directly contravening the GitHub community guidelines for posts?

(Tl;dr: the bug was valid, and is still valid.)
--------------------------------

Let’s ignore the threats, the mistakes and the misunderstandings plainly seen in those responses for a little bit. I’ll bring the conversation back on-topic before I respond to that threat – I shall assume some measure of good faith for the moment even though it’s clearly not in play from the maintainer of this fork of Manuel’s and Guido’s good work with generating the μSDX from the SDX design of Hans Summer (G0UPL / AF7BF).

Threatening a bug reporter is not a good look, and generally will not be accepted. I will temporarily tolerate this poor behaviour in order to make sure that my points are available and understood.

So:
Let's see if this reply can put forward in a way that's clear and understandable. It would be good that offence is not misguidedly taken at something imagined or misunderstood or non-existent. It would be preferable to ensure that the tools and information are now available to the maintainer to be fully aware of and can therefore address, the points being made about the currently extant issues with this fork of Guido’s code. 

I am however aware that I can only do so much. No matter how clearly things can be laid out in front of a person, it is never possible to understand things *for* that person.

Let’s try anyway, and it’ll be seen if I am to be again wantonly misunderstood.

Guido's code is GPLv3. Good to see this is labelled as such in this fork, as it must be. This is what affords the maintainer of this fork of Guido’s code the ability to **_distribute_** any changes in this fork to his (and others’) code. Let’s understand there is an emphasis on "**_distribute_**" - and this will be returned to later on. It is suspected that there’s a core disconnect here between the maintainer of this fork’s understanding of licenses, and the terms of the code so licensed.

In the readme.md here, it’s very interesting to see that there’s no links to anything Manuel (DL2MAN) has done regarding his work on improving Hans’ hardware design, that is needed to execute this fork of Guido’s code. The fact that the work of others _is_ linked, does give the reader a certain impression about things..

A more relevant item to this bug is the interestingly insidious disingenuous distortion of Guido's message from the top of the readme for his project:
The maintainer appears to have tried to reproduce a portion of Guido’s readme.md on this fork's readme, possibly in an attempt to not breach the licensing terms:

“Reproduced” from the original project:- 
> [blocks of text]
> “Note: there is an active online forum discussing the uSDX controlled by DL2MAN who has closed his own USDX branch (tr)usdx. Other groups on FB include "USDX USDR Radios", "(tr)USDX Open".”
> 73, Guido [pe1nnz@amsat.org](mailto:pe1nnz@amsat.org)

The trouble is - this is not what Guido wrote.

This is what Guido _actually_ wrote:

> uSDX is a simple and experimental (Class-E driven) SSB and CW SDR transceiver.
> [blocks of text]
> Note: there is an active online forum discussing the uSDX here: [uSDX Forum](https://groups.io/g/ucx).
> 73, Guido [pe1nnz@amsat.org](mailto:pe1nnz@amsat.org)

Can a difference be seen? Can the import of this difference, and the import of its location within Guido’s quote be understood?

It would appear that the misquote of Guido’s readme has the (hopefully unintended) consequence of suggesting that Guido expressly supports this particular fork of his code – a  suggestion which appears to be inaccurate at best. There does not appear to be any independent direct expression of Guido’s explicit support for this fork’s edits of his work. The statements that others can work on his code - subject to the licensing on that code - is not to be taken as an explicit support of this fork, as those statements are nothing more than expression of the licensing terms.

Editing/changing someone else's work and not attributing the changes correctly; this may be an impression gathered by those paying attention to the “work” of the maintainer of this fork, however valid or not that impression may be, it is still an impression that can be garnered.

(side note: I always support the idea that it's perfectly fine to build upon the work of others – but it’s _only_ perfectly fine when one obeys the licensing terms of the code and licensing terms of the hardware that one is working on, and attribute correctly what was received when starting, when distributing either the results of the code edits or distributing the actual code edits. )

Reading this misquote of Guido in this fork’s readme - it may give a reader an impression that there's something underhand and possibly even slimy going on here; even if there may be no real basis to that impression. It's not hard to subvert that impression - just be truthful and correct in what is said, and comply fully with the licenses already agreed to abide by when obtaining the code. It's not hard to do it right, but it is hard to restore the trust lost when a person finds out about the implications of the misquote.

It would be suggested that future responses refrain from threats to the bugreporter or others, and that responses stay on topic and only address the concerns raised. It would not be to the benefit of this fork to have the maintainer give such a po [... cut ...] I fear such a meeting though would unfortunately be to your detriment and to the amusement of myself and others.
</offtopic>

-----------------------

Back on-topic; the bug was valid, and is still valid.
-----------------------

Also – readers should always be very careful with reading tone from text. The reader takes the tone they expect from text, not that which was intended.. 
With this response I intend the tone to be accurate and neutral. Any other tone taken is at the risk of that reader.