GW8RDI/uSDXOpen #4: A6' was not declared in this scope

antonjan, 2023-06-05:
Hi
I think I had a look at all option and here is my description
Firstly I am not sure which version of the ino files to use I have the black brick. 
There is 4 files all all of them has the Black brick option

- RedCornersUSDX.ino
- WhiteButtonsUSDX.ino
- trusdx400c.ino
-usdx.ino)
Here is picture of my usdx unit (my understanding is it is called the black brick)
 link to image of my unit [https://www.giga.co.za/ocart/image/cache/data/QRP/USDX/HFQRPSDRTRANSC_4-500x500.JPG]
I have tried usdx.iso and RedCornerersUSDX.ixo
Enabling #define BLACK_BRICK 1
Disabling //#define RED_CORNERS 1 
It seems the problem is related to the SWR option but if I disabling the SWR option get the same error when compiling.

I am missing something. Could you possibly assist me in solving this problem.
I have tried version 1.8.10 Arduino as well as version version 2.1.0 getting the same error.

See error log below
Thanks Anton ZR6AIC

/home/anton/Downloads/usdx/usdx.ino: In function 'void initPins()':
usdx:400:17: error: 'A6' was not declared in this scope
 #define PIN_FWD A6
                 ^
/home/anton/Downloads/usdx/usdx.ino:5253:10: note: in expansion of macro 'PIN_FWD'
  pinMode(PIN_FWD, INPUT);
          ^~~~~~~
/home/anton/Downloads/usdx/usdx.ino:400:17: note: suggested alternative: 'AM'
 #define PIN_FWD A6
                 ^
/home/anton/Downloads/usdx/usdx.ino:5253:10: note: in expansion of macro 'PIN_FWD'
  pinMode(PIN_FWD, INPUT);
          ^~~~~~~
usdx:401:17: error: 'A7' was not declared in this scope
 #define PIN_REF A7
                 ^
/home/anton/Downloads/usdx/usdx.ino:5254:10: note: in expansion of macro 'PIN_REF'
  pinMode(PIN_REF, INPUT);
          ^~~~~~~
/home/anton/Downloads/usdx/usdx.ino:401:17: note: suggested alternative: 'AM'
 #define PIN_REF A7
                 ^
/home/anton/Downloads/usdx/usdx.ino:5254:10: note: in expansion of macro 'PIN_REF'
  pinMode(PIN_REF, INPUT);
          ^~~~~~~
/home/anton/Downloads/usdx/usdx.ino: In function 'void readSWR()':
usdx:400:17: error: 'A6' was not declared in this scope
 #define PIN_FWD A6
                 ^
/home/anton/Downloads/usdx/usdx.ino:5730:52: note: in expansion of macro 'PIN_FWD'
   v_FWD = v_FWD + (ref_V / 1023) * (int)analogRead(PIN_FWD);
                                                    ^~~~~~~
/home/anton/Downloads/usdx/usdx.ino:400:17: note: suggested alternative: 'AM'
 #define PIN_FWD A6
                 ^
/home/anton/Downloads/usdx/usdx.ino:5730:52: note: in expansion of macro 'PIN_FWD'
   v_FWD = v_FWD + (ref_V / 1023) * (int)analogRead(PIN_FWD);
                                                    ^~~~~~~
usdx:401:17: error: 'A7' was not declared in this scope
 #define PIN_REF A7
                 ^
/home/anton/Downloads/usdx/usdx.ino:5731:52: note: in expansion of macro 'PIN_REF'
   v_REF = v_REF + (ref_V / 1023) * (int)analogRead(PIN_REF);
                                                    ^~~~~~~
/home/anton/Downloads/usdx/usdx.ino:401:17: note: suggested alternative: 'AM'
 #define PIN_REF A7
                 ^
/home/anton/Downloads/usdx/usdx.ino:5731:52: note: in expansion of macro 'PIN_REF'
   v_REF = v_REF + (ref_V / 1023) * (int)analogRead(PIN_REF);
                                                    ^~~~~~~
exit status 1
'A6' was not declared in this scope

antonjan, 2023-06-05:
I eventually got the RedCornerUSDX.ino compiling  with Arduino Version 2.03. Still not sure if its the correct one to use for Black brick?
I did not enable Blackbrick or ReCorner kept the default setting.
I will try it in the meantime to see if it works and will give youfeedback
73 zr6aic

GW8RDI, 2023-06-05:
Hi, if you open the usdx.ino 4.00c file, and look down the start, you 
will see the define for black brick, just read my notes!  No one ever 
does of course lol!!

Best regards & 73 QuantumSDR Tea

Best regards & 73
Rob Colclough GW8RDI
*QuantumSDR Team*
QuantumSDR.com

Quantum SDR DSR product footer

Please click "Unsubscribe 
***@***.***?subject=Unsubscribe>" if you wish to 
be removed from future mailings.

antonjan, 2023-06-06:
If I enable black brick   #define BLACK_BRICK 1
I always get the same error as described above nothing else has change.
So it seems there is definitely something wrong?

GW8RDI, 2023-06-06:
No, nothing is wrong, I´m not sure what the error was, if it was not 
enough memory, you must disable some features, such as CW_messages_ext 
or disable CAT

Just read the instructions I left in the INO file, its all there!  Also, 
watch a youtube video on how to set the #define statements, it will help.

Best regards & 73 QuantumSDR Tea

Best regards & 73
Rob Colclough GW8RDI
*QuantumSDR Team*
QuantumSDR.com

Quantum SDR DSR product footer

Please click "Unsubscribe 
***@***.***?subject=Unsubscribe>" if you wish to 
be removed from future mailings.

antonjan, 2023-06-07:
Thanks
I have check 
//#define CW_MESSAGE 1          // Transmits pre-defined CW messages on-demand (left-click menu item 4.2)
//#define CW_MESSAGE_EXT 1      // Additional CW messages
is both disabled but same error. only cat is enabled
It seems the problem is related to the initPins() function
This only happens when I enable black brick
se below
/home/anton/Downloads/usdx/usdx.ino: In function 'void initPins()':
/home/anton/Downloads/usdx/usdx.ino:400:17: error: 'A6' was not declared in this scope

GW8RDI, 2023-06-07:
Okay, the error you are getting is caused by incorrect modifications to 
the #define XXXXX #ifdef -  #else - #endif statements.

Somewhere you have taken out half of one of these statements, leaving 
data and code unreachable.

You must fully understand how this works, there are some good videos on 
Arduino statement use on YouTube.

Best regards & 73 QuantumSDR Tea

  Otherwise go step by step using the original 4.00c file, change one 
thing at a time until you get the error, then look at what you last changed.

Best regards & 73
Rob Colclough GW8RDI
*QuantumSDR Team*
QuantumSDR.com

Quantum SDR DSR product footer

Please click "Unsubscribe 
***@***.***?subject=Unsubscribe>" if you wish to 
be removed from future mailings.

LUGEIX, 2023-11-01:
I got that Error with a ATmega386p board type. When I choose UNO, the error went away. Programmed worked great. Just need to get T37-6 Toroid to fix the 17, 12 and 10 meters issue. Figure how to turn the LED off. Fix the 10m motor boating issue. 
KA6VKP

chrisdebian, 2024-08-02:
Tidying up, can this be closed, now?

Chris
2E0FRU

BitLoose, 2024-08-02:
Sure

sysadminpower2019, 2025-07-23:
i'm getting similar errors with the red corners, even the file by itself with no modifications won't compile anymore