GW8RDI/uSDXOpen #5: Am appears double side band, alternative Am in code compile error

2E0WKS, 2023-06-21:
Hi have the radio working very well on FM etc with code provided.
The default AM appears to be double side band with no carrier.
It appears there is alternative method for better AM in source code as I understand it and tried to edit and IDE complaining about a
}
But line identified not obvious to why this happens.
Could you explain method to get AM carrier.

Thanks

BitLoose, 2023-07-26:
Hi, it's usually because the #define statement hasn't been correctly re-inserted, look for the #define and it's matching #endif, and get them to match.  Unfortunately the alternative AM code that I added doesn't help much, there seems to be a fundamental issue with AM, which is why Guido also hasn't got it working.  Lets not forget that our friendly little Arduino 328P is operated beyond specification, it is overloaded, the I2C is run at double the 400 KHz maximum so it's a small miracle it works at all.  On the other hand the problem may be fixed, but at the moment I just don't have the time.  One fix is to add the Spectrum DSP M2 to the uSDX, then you get all the modes properly decoded and a spectrum display along with it.

chrisdebian, 2025-07-03:
Hi, Daniel; do you know if this ever got resolved; can it be closed?

Chris
2E0FRU