GW8RDI/uSDXOpen #6: CAT_STREAMING

PE1DDA, 2023-06-29:
Hi Rob,

Some really nice modifications you made in 4.00c, especially for a SW-noob like myself :-)
Now I'm struggling with the "subject", Guido has released some software, for windows, that should make it possible to have only a single USB connection from uSDX to PC f.e. WSJT-X
You can find it here: https://github.com/threeme3/trusdx-audio
Basically it installs a version of Virtual Audio Cable and two virtual COM ports (Eterkit), so far so good, but, using your 4.00c with all necessaries enabled in the configuration section, I can't get it to work. Some CAT control in WSJT-X give a green but then Hamlib errors occur or timeouts etc. etc.  Any case: I can't get audio on the virtual cables.
Is that something you would like to have a look into it.......?
Warning: if you decide to install Guido's SW, be aware, it also installs a driver for uart CH340 and the version included is NOT the latest version (caused me some other troubles)

Much obliged!
Cees, PE1DDA

BitLoose, 2023-07-26:
Hi Cees, yes, I read the update on (tr)usdx, the programmer added an IQ stream for TX input from PC over VCom, the Open USDX code has the RX stream and I would have to add a TX stream to make it compatible, but to be honest I question it's worth, as I think Klarig does, given they aren't supporting it.
Personally, I would instead add a bluetooth or wifi audio connector, they cost around $12-14, and then no cables are needed at all. 
Connect your usdx or (tr)usdx to your PC via bluetooth for $4 while the offer lasts!  Forget cables, forget fiddling with an unsupported Virtual com streaming application!
https://www.temu.com/bt-adapter-audio-5-0-2-in-1-wireless-transmitter-receivertransmitter-with-3-5mm-cable-for-headphones-g-601099518792668.html?top_gallery_url=https%3A%2F%2Fimg.kwcdn.com%2Fproduct%2FFancyalgo%2FVirtualModelMatting%2F04e2d1126ac620d683b01b931366b6a6.jpg&spec_gallery_id=2005840876&refer_page_sn=10009&refer_source=10022&freesia_scene=2&_oak_freesia_scene=2&_x_vst_scene=adg&_x_ads_sub_channel=search&_x_ads_channel=bing&_x_ns_url=hfej0ZxK0P&_x_ads_creative_id=82669899152542&_x_ns_device=c&_x_ns_keyword=temu&_x_ads_account=176231188&_x_ns_match_type=b&_x_ns_msclkid=7ad949abe2c41380f2bd2d6aaa0edaec&_x_ads_set=518897482&_x_ns_source=o&_x_ads_id=1322714776571161&refer_page_name=search_result&refer_page_id=10009_1690388690184_sk004px1lg&_x_sessn_id=baxcloedhx

PE1DDA, 2023-07-27:
Hi Rob,

Thank you for looking into it and even more for your suggestion on bluetooth.
I think that is a bloody good idea (blame myself I never thought of that)!
73's de PE1DDA