GW8RDI/uSDXOpen #13: LSB and USB cannot be distinguished

sd8655131, 2024-03-15:
Hello, I have both trusdx and a China replica of usdx WHITEBUTTONS. Trusdx works perfectly, but when I control WHITEBUTTONS to switch between LSB and USB, there is no obvious change. I can even clearly receive FT8 in LSB mode. I've tried everything and can't solve this problem. Have you encountered this situation?

sajjsamm, 2024-03-21:
I encountered this problem several times. The solution is to replace the frontend passive components with low-tolerance and high-quality components. Replace all resistors with 1% type and all capacitors with at least 5% tolerance type with good quality like tantalum or film or polystyrene and avoid ceramic X5R or X7R or... at all cost.

chrisdebian, 2025-11-08:
Can this be closed, I wonder? Maybe add the info found, elsewhere.

Chris