GW8RDI/uSDXOpen #16: Boot error on RedCorner

mr-varga, 2024-07-06:
Hi, after flashing the 4.00d firmware. 
Often at startup the redcorner become freezed in a screen like in attached image, and remains fixed there making it necessary to restart.
[image]

drone777, 2024-07-10:
Try with connected charger or with external battery. If works good- change atmega fuses to lower watchdog voltage. Can't tell exactly as lost mine settings and don't want to disassemble.

GW8RDI, 2024-08-04:
The fault could be almost anything, hardware, bad upload, did you remove the 10nF capacitor connected to MOSI?  If not, firmware data can be corrupted.

n1zzo, 2025-02-19:
I am experiencing the same issue, and I did some investigation:
- The firmware corruption problem can be ruled out in my case as I read back the binary with avrdude and contained no errors
- I tried all the three brown-out voltage settings but the issue is still there
- I noticed that the issue appears only when turning on the radio after a few seconds, if you turn it on twice repeatedly, the second time almost always works

I am available to run further tests.