GW8RDI/uSDXOpen #18: power dropped

nazainurin, 2024-08-25:
before this, i got 5-6w using battery but after updtae this version im just got 1-3w in 40m and 20m band. 0.5-0.9w for 15m and 10m band. how i can fix it?

GW8RDI, 2024-08-25:
Your changes may have messed up filter selection or frequency.You're best off using the stock ino without changes.Post a picture of your usdx and the ino file you used.

nazainurin, 2024-08-26:
Im using usdxBLACKBRICK.uno
[image]

GW8RDI, 2024-08-26:
Well, 4.00d is working with hundreds of usdx none of which have low power, so you've got something bad, maybe a dry solder joint somewhere.You'll have to fault find.

Screenshot_20240826_095018_com.hihonor.photos.jpeg (view on web)