GW8RDI/uSDXOpen #21: problem with size code

tecnico56, 2024-09-26:
Hello a t all. I want to program the redcorners transceiver with the latest firmware version 4.00d. I use the Arduino 1.8.19 interface. When I start compiling the code it gives me the following error:

The sketch uses 31824 bytes (103%) of the available space for programs. The maximum is 30720 bytes.text section exceeds available space in board

Global variables use 1551 bytes (75%) of dynamic memory, leaving another 497 bytes free for local variables. The maximum is 2048 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
Error compiling for Arduino Nano board.
Thanks in advance

USBEprom, 2024-09-28:
@tecnico56 
Hi,
It could depend on the number of options you have activated in the sketch.
There is no enough space for all functions, you have to choose those you want by personalizing the Arduino sketch.
Example, you cannot have the active CAT function and all CW automatic messages at the same time , you must decide what is most important for you and go down to a compromise in order not to exceed the memory limit available in the microcontroller used in the uSDR+ RED CORNER, which is an ATMEGA328, then in Arduino IDE as board you should set "Arduino Uno", not "Arduino Nano" as you wrote here.

tecnico56, 2024-09-28:
> @tecnico56 Hi, It could depend on the number of options you have activated in the sketch. There is no enough space for all functions, you have to choose those you want by personalizing the Arduino sketch. Example, you cannot have the active CAT function and all CW automatic messages at the same time , you must decide what is most important for you and go down to a compromise in order not to exceed the memory limit available in the microcontroller used in the uSDR+ RED CORNER, which is an ATMEGA328, then in Arduino IDE as board you should set "Arduino Uno", not "Arduino Nano" as you wrote here. last but not least, It would be useful to use a more recent version of Arduino IDE, right now there is arduino-ide_2.3.2.

tecnico56, 2024-09-28:
> @tecnico56 Hi, It could depend on the number of options you have activated in the sketch. There is no enough space for all functions, you have to choose those you want by personalizing the Arduino [...]

closed 2024-09-28 by tecnico56

tecnico56, 2024-09-28:
@USBEprom. Thank you. I didn't realize that you have to choose between CAT or CW... how sad... Do you know a fw version without the CAT management code? Thanks

GW8RDI, 2024-09-29:
No choice between cw and cat, but therec are many options that need fine tuning, as i wrote in the ino file

USBEprom, 2024-09-29:
@tecnico56

Hi,
there is no a firmware version without the CAT management code, the firmware is always **usdxREDCORNERS.ino** (https://github.com/GW8RDI/uSDXOpen/blob/main/usdxREDCORNERS.ino), it simply must be configured for your needs.
CAT feature is in line **228**, you need to disable it by add a double-slash at line-start, as in the following example:

**228 //#define CAT              1   // CAT-interface - OTHER OPTIONS, SUCH AS CW_MESSAGES and KEEP_BAND_DATA MAY TO BE DISABLED TO MAKE SPACE FOR CAT**

But as Rob GW8RDI wrote, **there are many options that need to fine tuning**, as he clearly explained at the top of the sketch **usdxREDCORNERS.ino**.
How to enable/disable a feature is explained in line **96** of the sketch:

**96 // Configuration switches; remove/add a double-slash at line-start to enable/disable a feature; to save space disable e.g. CAT, DIAG, KEYER**

In the lines of the sketch there is information of all kinds to optimize the code, it is only need to read everything carefully and stick to what is reported there.
It is not difficult, in the sketch everything is explained very well.
In my case, following the indications I customized **usdxREDCORNERS.ino** by disabling **CAT** (line 228) and **KEEP_BAND_DATA** (line 151) features which do not interest me and in their place I have activated automatic CW messages and further features, so that my uSDR+ RED CORNERS now has 6 preset automatic CW messages, FAST AGC, SWR, MOX and more.
Everything works like a charm and I am very happy with it.
**For all this I thank Rob GW8RDI for the terrific sketch he wrote and shared so that anyone can benefit from it, thank you Rob!**

GW8RDI, 2024-09-29:
Thank you!  It's good to know someone made good use of my notes!Your configuration sounds excellent, if it also has CAT enable, please share it with the group.73, Rob GW8RDI

tecnico56, 2024-09-30:
@ USBEprom : Thank you very much for your valuable information. In fact, my interest is to have the redcorners with the firmware 4.00d set only for LSB/USB voice and CW. I don't want the digital modes. However, it would be nice to do some cleaning in the firmware and compress the code to fit in the operating parameters of the Mega328P. But this is for the firmware masters and it's another story for me who only understands hardware.

tecnico56, 2024-10-01:
@USBEprom  -  Thank you for your reply. You are right, it takes a lot of attention to make the changes, in fact thanks to your suggestions, I was able to customize my redcorners. Commenting on the defines that are not of interest to me, I noticed that the space occupied by the sketch occupies 93% compared to the original sketch that occupies 103% of the Mega328P memory. So the problem of the error given by the Arduino compiler has also been eliminated. Now my transceiver is suitable for voice and CW even if I do very little CW. Hi! Speaking of voice: I noticed that the microphone capsule mounted on board is qualitatively poor. I replaced it with a micro capsule, like the microphones of cell phones, which have a better bandwidth for narrow band voice. A second modification that greatly improved the modulation in voice, was to replace the 100nF capacitor that carries the audio signal from the microphone to pin 16 (MISO) of Arduino, with a 1uF ceramic smd 0603. That's all for now. Best 73.