GW8RDI/uSDXOpen #31: CAT Error with WSJT-X

n1zzo, 2025-03-16:
I flashed the latest version (9ac76129a8c840453147d2fc06727c825e832b81) of uSDXOpen, and I enabled the CAT interface over UART with 115k baud rate.
I am using the radio with WSJT-X on Raspberry Pi 400 with PostmarketOS.
After a couple of hour of operation the CAT interface reports an error:

```
Hamlib error: kenwood_transaction: No data expected, checking ID; in
kenwood_transaction: wrong reply kenwood_transaction: cmdstr = FA00014095600
read_string_generic called, rxmax=4095 direct=1, expected_len=1
write_block(): TX 14 bytes
0000    46 41 30 30 30 31 34 30 39 35 36 30 30 3b           FA00014095600;
write_block(): TX 3 bytes
0000    49 44 3b                                            ID;
read_string_generic called, rxmax=51 direct=1, expected_len=1
read_string_generic(): RX 8 characters, direct=1
0000    00 00 49 44 30 32 30 3b                             ..ID020;
kenwood_transaction: read_string len=0 ''
kenwood_transaction: No data expected, checking ID; in
kenwood_transaction: wrong reply kenwood.c(711):kenwood_transaction returning2(-8) Protocol error

kenwood.c(2012):kenwood_set_freq returning2(-8) Protocol error

**2:rig.c(2238):rig_set_freq returning(-8) Protocol error

Protocol error
Protocol error
 while setting frequency

Timestamp: 2025-03-15T21:09:55.935Z
```

Here is another one:

```
Hamlib error: kenwood_transaction: No data expected, checking ID; in
kenwood_transaction: wrong reply kenwood_transaction: cmdstr = FA00021094600
read_string_generic called, rxmax=4095 direct=1, expected_len=1
write_block(): TX 14 bytes
0000    46 41 30 30 30 32 31 30 39 34 36 30 30 3b           FA00021094600;
write_block(): TX 3 bytes
0000    49 44 3b                                            ID;
read_string_generic called, rxmax=51 direct=1, expected_len=1
read_string_generic(): RX 7 characters, direct=1
0000    00 49 44 30 32 30 3b                                .ID020;
kenwood_transaction: read_string len=0 ''
kenwood_transaction: No data expected, checking ID; in
kenwood_transaction: wrong reply kenwood.c(711):kenwood_transaction returning2(-8) Protocol error

kenwood.c(2012):kenwood_set_freq returning2(-8) Protocol error

**2:rig.c(2238):rig_set_freq returning(-8) Protocol error

Protocol error
Protocol error
 while setting frequency

Timestamp: 2025-03-16T02:13:55.843Z
```

Could you give me some pointers on where to look?

Thanks!

IU2KIN

n1zzo, 2025-03-16:
I'm seeting the error on hamlib 4.0.6, will try to build WSXT-X with the latest version of hamlib and see if the issue persists.

GW8RDI, 2025-03-17:
If it works for 2 hours and resetting the Spectrum doesn't fix it, try 19200 baud.Best regards, 73Julian EvansQuantumSDR TeamQuantumSDR.com

I flashed the latest version (9ac7612) of uSDXOpen, and I enabled the CAT interface over UART with 115k baud rate.

n1zzo, 2025-03-17:
Setting the baud rate to 19200 indeed solves the intermittent errors with Red Corner USDX.
Thanks for the help!

IU2KIN

n1zzo, 2025-03-18:
I've enabled radio Tx in WSJT-X and after some hours this error appeared:

```
Hamlib error: kenwood_transaction: read_string len=2 '?;'
kenwood_transaction: Unknown command or rig busy 'IF'
read_string_generic called, rxmax=128 direct=1, expected_len=1
read_string_generic(): RX 2 characters, direct=1
0000    3f 3b                                               ?;
kenwood_transaction: read_string len=2 '?;'
kenwood_transaction: Unknown command or rig busy 'IF'
kenwood.c(711):kenwood_transaction returning2(-9) Command rejected by the rig

kenwood.c(763):kenwood_safe_transaction returning2(-9) Command rejected by the rig

****4:kenwood.c(1243):kenwood_get_if returning(-9) Command rejected by the rig

***3:kenwood.c(5087):kenwood_get_ptt returning(-9) Command rejected by the rig

**2:rig_get_ptt: elapsed=108ms
**2:rig.c(3947):rig_get_ptt returning(-9) Command rejected by the rig

Command rejected by the rig
Command rejected by the rig
 while getting PTT state

Timestamp: 2025-03-18T00:05:55.985Z
```

I have ferrite chokes on my UART cables, could this be a software issue?

n1zzo, 2025-03-26:
I've performed a capture of the serial interface, the issues were not related to the baud rate but to a missing implementation of the CAT FW command that retrieves the filter settings. WSJT-X queries this commands very often and the radio returns error every time. After a couple of hours of operation two error responses appear in the Tx buffer, crashing hamlib.
I've implemented the fix, tested on my radio for a couple of days without crashes and opened a PR: https://github.com/GW8RDI/uSDXOpen/pull/32.

Please review it and let me know if it's good to be merged.

Best regards,

IU2KIN

sysadminpower2019, 2025-07-03:
I had a similar issue maybe its best to just fork the repo and implement the missing settings

sysadminpower2019, 2025-07-10:
Couldn't you just fork the repo and do it?

n1zzo, 2025-07-20:
Of course, although right now I don’t have enough time so I’ll keep my fork of this repo updated with my fixes to the CAT.

sysadminpower2019, 2025-07-23:
where is your fork?

n1zzo, 2025-08-04:
> *sysadminpower2019*
>  left a comment
> (GW8RDI/uSDXOpen#31)
> where is your fork?

Sorry for the long reply,
here it is:

<https://github.com/GW8RDI/uSDXOpen/pull/32>

Thanks,

Niccolò Izzo IU2KIN

GW8RDI, 2025-08-31:
There is a base usdx.ino file, which has all the switches apart from the trusdx version, which is currently 4.00c.